Rahix/FusedFilamentDesign #3: Do you wan to make this into a workbench that is downloadable through the Addon Manager ?

luzpaz, 2025-05-13:
More people could benefit from it and the Hole function could get more exposure and updated in FreeCAD core as a consequence ?

https://github.com/FreeCAD/FreeCAD-addons?tab=readme-ov-file#adding-your-workbench-to-the-collection  
(you can skip steps 1 and 2)

Rahix, 2025-05-13:
Yes, I very much planned on making this available through the addon manager! :)  I wanted to sit on it for a bit more time, to iron out the worst issues and add more of the features I have in mind, that's why I didn't announce it anywhere yet.  Will get on that soon!

> Updated in FreeCAD core as a consequence

Yeah, I was also thinking about this.  I think in the long run, it is a good idea, but for the time being, these sort of specialty features can evolve much better as an external project that isn't coupled to the upstream release schedule.  Once the features have settled and we have some data on what has proven itself and what hasn't, things can move upstream.

And sort of related - I specifically designed this addon so it generates FreeCAD core objects instead of using FeaturePython.  The intention is that this keeps the models built using this addon editable on FreeCAD installations where the addon is not installed.  I'm not sure how much others care about this, but for me at least it is important to keep my models easily redistributable.

Rahix, 2025-05-15:
Submitted: https://github.com/FreeCAD/FreeCAD-addons/pull/379 :)

closed 2025-05-16 by Rahix