Rahix/FusedFilamentDesign #15: Not possible to select just certaint holes

pierreporte, 2025-06-09:
`ffDesign_HoleWizard` takes a PD hole feature as input, which means that all the holes created by it will have the dress-up feature. It is not possible to apply it only to a subset.

Related: #14.

Rahix, 2025-06-10:
This is inspired by the way `PartDesign` features work.  You also can't use a subset of a sketch for the `PartDesign_Hole` feature, for example.  If you want to distinguish, create two sketches and two `Hole` features.

Or, if you really want to, just edit the sketches/features created by the `ffDesign_` commands and delete everything for the holes you don't want to apply to.

hyarion, 2025-06-10:
> I don't quite see the usecase here... 

The ceiling angle depends on the printer quality and is usually defined as `overhang threshold` in the printer settings in the slicer.
By specifying how vertical holes and arcs (maybe other surfaces as well?) should be cared for using dressups you can easily create and distribute a model which the user can adjust to be printed.

It could be (in some iterations down the line) as easy as loading your slicer's printer settings in this addon and everything would auto-adjust for your usecase.

> For shapes like yours, why not simply model the corner into the sketch to begin with?

You can model corners into holes too if you change it to pockets, but arguing that pretty much invalidates the purpose of this nice addon. I think you've done something really great here and I think this could be extended into something amazing.

> This will be more flexible and more robust in every way, than a dressup feature on top.

I got carried away a bit, but this doesn't need to be handled with selecting faces, this case could also be handled with allowing any feature to be used as base (pocket, pad, revolve can all create round overhangs). But you might also want to handle overhangs differently depending on where they are on the model. Holes (etc) which are close to a top surface might be too close to have a complete teardrop shape, there you might want to opt to a top bridge or even a more aggressive angle which slightly droops. You could argue that the addon can try to be smart about this, but providing good defaults and allowing users to fine tune settings is usually preferred (at least in my experience). In this case that could mean on a overhang by overhang basis.

Rahix, 2026-11-06:
Slightly tangential to this topic: As of #57 (released in version 0.26.360), the addon correctly generates dress-up features for `PartDesign_Hole`s which were built from partial selection of a sketch rather than a full sketch.