Rahix/FusedFilamentDesign #17: Teardrop shape and roof bridge should be one feature, with more parameters

pierreporte, 2025-06-09:
The teardrop shape and roof bridge are very similar. Basically roof bridge is a teardrop shape with a zero-length bridge. The two modes could be merged into one, with the bridge length being a user parameter.

Also, the angle could be a user parameter as well.

pierreporte, 2025-06-10:
Indeed, the sketch would need to change when switching to one mode to another.

We could argue that they are different and the choice is made early, but there could be cases when the designer would need to switch type as design evolves. IMO they are extremely similar.

The angle is indeed a property but FreeCAD tries to rely less and less on the property panel, and allowing the user to edit values directly within the task panel.

Rahix, 2025-06-12:
> need to switch type as design evolves

You can always remove the feature and then add the other kind.  I think that's reasonable effort for the amount of times it will be necessary.

In fact, the same "problem" is present with standard chamfers vs. fillets.  And I would argue, it is a lot more cumbersome there, because you have to re-select all the edges when switching from chamfer to fillet.

> The angle is indeed a property but FreeCAD tries to rely less and less on the property panel, and allowing the user to edit values directly within the task panel.

It would make sense to have it in the task panel if ffDesign had a preview of the feature that is about to be generated.  But without a preview, you won't really be able to guess the right angle before the feature is actually there.  That's why I did not include the angle in the task panel.