Rahix/FusedFilamentDesign #21: Implement sacrificial layers

pierreporte, 2025-06-09:
[Rule 3.4](https://blog.rahix.de/design-for-3d-printing/#rule-R3.4) from the article written by @Rahix

Sacrificial layers are very useful to reduce the need for support. This addon should provide a way to make them. The user must be able to specify layer height.

redPanther, 2025-06-11:
For my understanding for this you just select the edges of the hole and then press the pad button, type 0.2 and ready - or just reduce the hole depth by - 0.2 in the first place. How can this plug in speedup the process? Did I miss something?

pierreporte, 2025-06-11:
@redPanther it is not necessary a hole and the zone to fill with a sacrificial layer may not be circular.

Rahix, 2025-06-13:
What @redPanther says is indeed true.  You can select an arbitrary loop of edges on a part and then pad by 0.2mm to make a sacrificial layer.  The downside of this approach is that the resulting feature will of course be very geometry dependant.

I think the only useful tool we could still provide on-top would be one that takes a pocket feature and creates a pad right at the bottom of the pocket. The pad then serves as a sacrificial layer for supporting any holes reaching further down.