Rahix/FusedFilamentDesign #22: Teardrop shape and roof bridge direction can’t be set along existing geometry

pierreporte, 2025-06-09:
According to the documentation, those feature can be oriented with a variable, even though it’s not visible either in the task panel nor the property view. The user then must know the exact angle to enter. It is not possible to select existing geometry or datum to orient the feature correctly.

Rahix, 2025-06-10:
> It is not possible to select existing geometry or datum to orient the feature correctly.

I am personally not at all a fan of using geometry references, because they can be fragile.  Also, I don't quite see how a geometry reference for the angle would help much.  An edge could be a reference of the printing direction, but this would still be ambiguous as the edge can be interpreted two ways.

I did consider adding a way to encode the print orientation of a Body as a sort of body-global property, but ultimately decided not to pursue this as it adds a lot of complexity for only little gain.

pierreporte, 2025-06-11:
The settings object (that could be a VarSet) is the solution proposed by the FreeCAD Design Working Group for the sheet metal workbench to contain common values like thickness and bending radius (that can individual features could override). It hasn't been implemented yet and the discussion isn't over, but it would be interesting if there was a common solution. Ease of use is also consistency across the program and every add-on that diverges from FreeCAD UI/UX is harder to use as a consequence.

It is of course understandable to keep the add-on simple while it's still new.