Rahix/FusedFilamentDesign #26: I don't see FusedFilamentDesign in FreeCad addons manager

DanielMazurkiewicz, 2025-07-19:
I may be probably missing something, but I don't see this addon in list of addons. Diogging further, I don't see it here as well:

https://github.com/FreeCAD/FreeCAD-addons

(using FreeCad 1.0.1)

Rahix, 2025-07-19:
Hi,

interesting.  Unfortunately, I am not able to reproduce the issue, it shows up fine for me...  Did you refresh the addon cache?  Did you accidentally enable a filter in the addon manager (make sure it is set to `Filter: Workbench, Any`)?

The addon doesn't show up as a checked-out module in the addons repo, but it is still part of the catalog: https://github.com/FreeCAD/FreeCAD-addons/blob/29d47461679b168ea0c368de75f669ccc48477f6/AddonCatalog.json#L603-L610

DanielMazurkiewicz, 2025-07-24:
Well, it was add-on cache, refreshing it solved problem, sorry for bothering and thank you for tips!

closed 2025-07-24 by DanielMazurkiewicz