Rahix/FusedFilamentDesign #27: Hole Wizard does not like non-construction External Geometry.

BackwardEcho, 2025-07-26:
At least for the Thread Forming Ribs, the process breaks when the selected hole is created using non-construction (solid) line projected external geometry. It works fine with construction (dashed) line geometry with a normal circle on top of it though.

[image]

[image]

[image]

And the FC version info it it helps at all:

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/07/24 17:43:23
Build type: Release
Branch: main
Hash: 45adb828a6652c272c0b215fe112f6ca763cd0b1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/32.512/1.25
Installed mods: 
  * AddonManager 2025.7.17
  * fasteners 0.5.39
  * freecad.gears 1.3.0
  * FusedFilamentDesign 0.25.250
  * Telemetry 1.0.2

...and a side note since I don't know if it's a problem but I'm not sure where else to ask: Is that Rib Thread Template supposed to not have a proper attachment?

Rahix, 2025-07-27:
I can reproduce.  The problem is that we only look at `Sketch.Geometry` but not `Sketch.ExternalGeo`.  Unfortunately, I could not easily figure out how to read the defining/reference property for geometry elements in `Sketch.ExternalGeo`?  For `Sketch.Geometry` we have `Sketch.getConstruction()` but no similar method seems to exist for `ExternalGeo`...

> ...and a side note since I don't know if it's a problem but I'm not sure where else to ask: Is that Rib Thread Template supposed to not have a proper attachment?

Yes, this is sort of intentional.  The placement of the template sketch does not matter because it is just a template which is cloned (via shapebinders) to all the actual thread locations.  Thus the template itself does not need to be attached to anything.

BackwardEcho, 2025-07-27:
> I can reproduce. The problem is that we only look at `Sketch.Geometry` but not `Sketch.ExternalGeo`. Unfortunately, I could not easily figure out how to read the defining/reference property for [...]

Considering a hole, pocket, or pad can be made with the non-construction external geometry alone, are you able to see how those operations treat said external geometry and utilize their methods for here?

Not sure what else to add regarding this topic as you have labeled it as a bug, so I'm assuming you want to make it work with non-construction external geometry.

> Yes, this is sort of intentional. The placement of the template sketch does not matter because it is just a template which is cloned (via shapebinders) to all the actual thread locations. Thus the template itself does not need to be attached to anything.

To be honest, it feels wrong to have a sketch not attached to anything. To make it feel less wrong, I placed the rib sketches (that I have used so far) within the tree / drop down menu for the specific part they are used for even if they are still unattached. It at least keeps things tidy in my project file.

Rahix, 2025-07-28:
> Considering a hole, pocket, or pad can be made with the non-construction external geometry alone, are you able to see how those operations treat said external geometry and utilize their methods for here?

I have not checked, but from past experience diving into the PartDesign tools: They use direct access via C++ APIs.  A lot of that is not exposed well to the Python API, making it hard to reproduce exact behavior.

> Not sure what else to add regarding this topic as you have labeled it as a bug, so I'm assuming you want to make it work with non-construction external geometry.

Yeah, would be nice to fix this.  Generally, the HoleWizard tools should just mirror the PartDesign_Hole behavior as precisely as possible.

pierreporte, 2025-08-03:
@PaddleStroke implemented the new external geometry. Maybe he can have relevant input for this problem.