Rahix/FusedFilamentDesign #28: FusedFilamentDesign  missing from Addon Manager

tre4b, 2025-07-28:
FusedFilamentDesign  is not showing up in the Addon manager.

Not sure why it just does not show up at all.

Rahix, 2025-07-28:
Duplicate of #26 maybe?

Did you refresh the addon cache?

Did you accidentally enable a filter in the addon manager (make sure it is set to Filter: Workbench, Any)?

Rahix, 2025-07-29:
Huh, interesting.  For reference, we specify the license here:

https://github.com/Rahix/FusedFilamentDesign/blob/0ff3bc9768e8e78d49aff8d9abdef193764d6045/package.xml#L9

That's the same license used by FreeCAD itself if I recall correctly...

Rahix, 2025-07-31:
Closing this for now, but we can reopen the issue if more people stumble into this problem.

closed 2025-07-31 by Rahix